esrlabs/northstar #66: List required Kernel features

marcmo, 2020-07-28:
We already know what kernel features are required in order to run northstar.
We should list them so porting will become easier.

maybe something like this:
ok…dann wenn der Peter wieder da ist…vielleicht in so einer Form:
**DM-Verity support**:
```CONFIG_DM_VERITY=m
CONFIG_DM_VERITY_VERIFY_ROOTHASH_SIG=y
CONFIG_FS_VERITY=y
CONFIG_FS_VERITY_BUILTIN_SIGNATURES=y
```

**Namespace Support**:
…

some of the features required are probably:
```
CONFIG_DM_VERITY=m
CONFIG_DM_VERITY_VERIFY_ROOTHASH_SIG=y
CONFIG_FS_VERITY=y
CONFIG_FS_VERITY_BUILTIN_SIGNATURES=y
CONFIG_SQUASHFS=y
CONFIG_SQUASHFS_FILE_DIRECT=y
CONFIG_SQUASHFS_DECOMP_SINGLE=y
CONFIG_SQUASHFS_XATTR=y
CONFIG_SQUASHFS_ZLIB=y
CONFIG_SQUASHFS_LZ4=y
CONFIG_SQUASHFS_LZO=y
CONFIG_SQUASHFS_XZ=y
CONFIG_SQUASHFS_ZSTD=y
CONFIG_SQUASHFS_FRAGMENT_CACHE_SIZE=3
CONFIG_LOOPBACK_TARGET=m
CONFIG_RC_LOOPBACK=m
CONFIG_PID_NS=y
...
```

bzld, 2020-07-28:
FS_VERITY is something different than DM_VERITY...

First - unverified - draft of the list:
```
CONFIG_CGROUP_PIDS=y
CONFIG_CGROUP_FREEZER=y
CONFIG_CGROUP_DEVICE=y
CONFIG_CGROUP_CPUACCT=y
CONFIG_CGROUP_PERF=y
CONFIG_CGROUP_BPF=y
CONFIG_CGROUP_DEBUG=y

CONFIG_NAMESPACES=y
CONFIG_USER_NS=y

CONFIG_DEVTMPFS=y
CONFIG_DEVTMPFS_MOUNT=y

CONFIG_BLK_DEV_LOOP=y
CONFIG_BLK_DEV_LOOP_MIN_COUNT=128 (value might not need to be this high)
CONFIG_AIO=Y (potentially optional)

CONFIG_CRYPTO_SHA256=Y
CONFIG_DM_VERITY=y

CONFIG_SQUASHFS=y
CONFIG_SQUASHFS_FILE_DIRECT=y (potentially optional)
CONFIG_SQUASHFS_DECOMP_MULTI=y (recommended, but optional)
CONFIG_SQUASHFS_XATTR=y (potentially optional)
CONFIG_SQUASHFS_ZLIB=y (may be removed if we switch to zstd-only)
CONFIG_SQUASHFS_ZSTD=y
```

pclesr, 2020-08-11:
CONFIG_CGROUPS
        CONFIG_MEMCG
        CONFIG_BLK_CGROUP
        CONFIG_CGROUP_SCHED
        CONFIG_CGROUP_PIDS
        CONFIG_CPUSETS
        CONFIG_CGROUP_DEVICE
        CONFIG_CGROUP_CPUACCT
        CONFIG_CGROUP_PERF

CONFIG_NAMESPACES
        CONFIG_UTS_NS
        CONFIG_USER_NS
        CONFIG_PID_NS
        CONFIG_NET_NS
        CONFIG_IPC_NS

CONFIG_BLK_DEV_DM
        CONFIG_DM_UEVENT
        CONFIG_DM_CRYPT
        CONFIG_DM_THIN_PROVISIONING
        CONFIG_DM_VERITY
        CONFIG_DM_VERITY_FEC

CONFIG_BLK_DEV_LOOP
        CONFIG_BLK_DEV_LOOP_MIN_COUNT=64

CONFIG_CRYPTO_SHA256

CONFIG_TMPFS

CONFIG_SQUASHFS
        CONFIG_SQUASHFS_FILE_DIRECT
        CONFIG_SQUASHFS_DECOMP_MULTI
        CONFIG_SQUASHFS_4K_DEVBLK_SIZE
        CONFIG_SQUASHFS_XATTR
        CONFIG_SQUASHFS_ZLIB
        CONFIG_SQUASHFS_LZ4
        CONFIG_SQUASHFS_LZO
        CONFIG_SQUASHFS_XZ
        CONFIG_SQUASHFS_ZSTD
        CONFIG_SQUASHFS_EMBEDDED
                CONFIG_SQUASHFS_FRAGMENT_CACHE_SIZE=3

pclesr, 2020-08-11:
For network namespace config:
CONFIG_BRIDGE
CONFIG_BRIDGE_VLAN_FILTERING
CONFIG_MACVLAN
CONFIG_MACVTAP
CONFIG_VXLAN
CONFIG_VETH

CONFIG_IP_NF_IPTABLES
        CONFIG_IP_NF_NAT
        CONFIG_IP_NF_TARGET_MASQUERADE
        CONFIG_IP_NF_MANGLE
        CONFIG_IP_NF_FILTER
        CONFIG_IP_NF_SECURITY

flxo, 2020-09-17:
Think @pclesr wrote a script that check for the options on a running system. Can we move that into the workspace and leave a hint in the docs about this?

flxo, 2020-09-17:
Ah - there's a dedicate ticket #34

closed 2020-09-17 by flxo