esrlabs/northstar #101: Check minijail mount settings

bzld, 2020-09-28:
TODO: Check we have sensible, locked-down mount settings by running `mount` within a container and check the output.
For example:
- Everything `nosuid`
- Everything `nodev` except /dev
- Everything except read-only root fs `noexec`
- /proc and /sys as `ro`

libminijail has specialzied `mount_dev`, `mount_tmp(_size)` calls which may have good defaults built-in.

flxo, 2020-10-09:
#111 uses `mount_dev` and `mount_tmp`.

Nosuid needs to be done

flxo, 2020-11-13:
SUID should be a optional flag fro bind mounts.

flxo, 2021-05-19:
The `inspect`container outputs the containers mount table which is typically something like

minijail:
```
2021-05-19T08:39:11.991Z: DEBUG: runtime::minijail inspect: /proc/self/mounts:
2021-05-19T08:39:11.991Z: DEBUG: runtime::minijail inspect:   /dev/dm-0 / squashfs ro,nosuid,nodev,relatime 0 0
2021-05-19T08:39:11.991Z: DEBUG: runtime::minijail inspect:   minijail-devfs /dev tmpfs rw,nosuid,noexec,relatime,size=5120k,mode=755 0 0
2021-05-19T08:39:11.991Z: DEBUG: runtime::minijail inspect:   none /tmp tmpfs rw,nosuid,nodev,noexec,relatime,size=20000k 0 0
2021-05-19T08:39:11.991Z: DEBUG: runtime::minijail inspect:   /dev/sda1 /lib ext4 ro,relatime,errors=remount-ro 0 0
2021-05-19T08:39:11.991Z: DEBUG: runtime::minijail inspect:   /dev/sda1 /lib64 ext4 ro,relatime,errors=remount-ro 0 0
2021-05-19T08:39:11.991Z: DEBUG: runtime::minijail inspect:   proc /proc proc ro,nosuid,nodev,noexec,relatime 0 0
```

island:
```
2021-05-19T08:40:09.206Z: DEBUG: runtime::island::io inspect: /proc/self/mounts:
2021-05-19T08:40:09.206Z: DEBUG: runtime::island::io inspect:   /dev/dm-0 / squashfs ro,nosuid,nodev,relatime 0 0
2021-05-19T08:40:09.206Z: DEBUG: runtime::island::io inspect:   proc /proc proc ro,relatime 0 0
2021-05-19T08:40:09.206Z: DEBUG: runtime::island::io inspect:   udev /dev devtmpfs rw,nosuid,relatime,size=4062420k,nr_inodes=1015605,mode=755 0 0
2021-05-19T08:40:09.206Z: DEBUG: runtime::island::io inspect:   none /tmp tmpfs rw,nosuid,nodev,noexec,relatime,size=20000k 0 0
2021-05-19T08:40:09.206Z: DEBUG: runtime::island::io inspect:   /dev/sda1 /lib ext4 ro,nosuid,nodev,relatime,errors=remount-ro 0 0
2021-05-19T08:40:09.206Z: DEBUG: runtime::island::io inspect:   /dev/sda1 /lib64 ext4 ro,nosuid,nodev,relatime,errors=remount-ro 0 0
````

flxo, 2021-05-31:
Setting the mount flags could be done for binds in the manifest

```rust
#[derive(Clone, Eq, PartialEq, Debug, Hash, Serialize, Deserialize)]
pub enum MountFlag {
    /// Bind mount
    #[serde(rename = "rw")]
    Rw,
    // Mount noexec
    // #[serde(rename = "noexec")]
    // NoExec,
    // Mount noexec
    // #[serde(rename = "nosuid")]
    // NoSuid,
}
```

Probably the logic should be inverted to explicitly enable `exec` `suid` and `dev`.

PR #359 (2021-06-17): Mount option cleanup
Extend the manifest with mount options for `resources` and `binds`. Adjust the `proc` mount options.

Fixes #101
commit 9df694ab (2021-06-17): Add mount options

Add mount options to manifest for bind, resource and adjust proc mount
options.

Fixes #101
closed 2021-06-17 by flxo