esrlabs/northstar #160: Container directories: installation and read writeable

flxo, 2020-11-03:
Currently the installation path picks the first container dir in the configuration. This implies that this one is writeable and not very intuitive.

Extend the configuration with a writeable container dir that is used for the installs. This may be a single one? The list of ro containers dirs might remain.

marcmo, 2020-11-04:
I think it makes sense to have a single writable path for installations, everything else will be too complicated.
Not sure why we need the feature to have multiple registries. What about the uninstallation? for that we also need `write` permissions on the folder. 
would only make sense if we explicitly want to disable messing with some of the installed applications.

flxo, 2020-11-13:
Superseded by #174

closed 2020-11-13 by flxo