esrlabs/northstar #170: Support using northstar npk filesystem for firecracker

marcmo, 2020-11-10:
Filesystem Image handling: 
Verify if firecracker can pass through a `dm verity` block device to the VM. The goal is to make the container FS image available to the VM

pclesr, 2020-11-11:
Update that I posted to slack:

To make testing of mounting the block devices easier, I changed the mount propogation flags in `north/src/runtime/linux/mod.rs` from MS_PRIVATE to MS_SHARED.  After building the java MOD` /system` directory as a north container, then I could easily see where things are mounted, such as:
```
/dev/block/dm-1 on /data/local/tmp/north/run/fc_java_mod/0.0.1 type squashfs (ro,seclabel,relatime)
/dev/block/dm-2 on /data/local/tmp/north/run/hello/0.0.2 type squashfs (ro,seclabel,relatime)
```
If I configure the VM to have` /dev/block/dm-1` as one of it's disks, and a file on the host as it's second disk (for `/data`), I can then mount and run the java app in the VM.

Mounted disks :
`/ `(initrd) :   binaries and libraries for java, using musl openjdk-11
`/system` (/dev/block/dm-1)  northstar container with the MOD jar files, mounted via _`mount -t squashfs -o ro /dev/vdb /system`_ 
`/data` (just a filesystem image file on the host)

Summary: it is possible to mount the verity device in the VM, so it should be reasonably straightfoward to configure north to either mount the block device on the host (for normal containers) or to pass it to the VM.

flxo, 2022-05-02:
Archive fc issues

closed 2022-05-02 by flxo