esrlabs/northstar #171: Develop protocol for passing configuration information from host to guest over vsock

marcmo, 2020-11-10:
- 	container manifest
- 	network setup information
- 	any additional setup information (environment variables, logging info, ??? )

pclesr, 2020-11-11:
The documentation for vm <--> host communication for firecracker (FC) is here:` https://github.com/firecracker-microvm/firecracker/blob/master/docs/vsock.md` . It recommends using both `Socat` and `nc-vsock.` 

I compiled both statically for ARM, and using `Socat` on the host as the server and `nc-vsock` in the VM as the client, it is possible to send a configuration file at boot time from the host to the guest. This eliminates the need to specify the IP params on the kernel config.

To send a manifest or config file, the host sets up a Socat process listening on a socket:
`	socat $file UNIX-LISTEN:${dir}/fcvsock.sock_${port}`
The` fcvsock.sock `socket is specified in the firecracker config file and is created by the FC process. The socat server should be started before FC (to avoid races), so it needs to wait until FC creates the socket. The port number is arbitrary and analogous to a normal TCP port (it is used to distinguish between processes on the host).

The client (VM) can then use nc-vsock to connect to the host and pull the file:
`	/bin/nc-vsock 2 $port > $outfile`
The port number is currently specified on the kernel command line. If the vsock is placed in a unique directory on the host, then all VMs can use the same port, which would eliminate the need to modify the kernel command line on each VM invocation.

Using this method, it is possible to pass the network config from host to guest; the guest then invokes the appropriate `ip` commands to configure eth0.

I have not yet investigated using Socat as a pipe, but it is theoretically possible. This would mean that all logging in the guest would be directed to a socket (vsock) in the VM; in the host, a process would continually drain the pipe and write the results to the logger.

It might be worth having a few dedicated sockets, in a manner similar to stdin/stdout/stderr. One socket for configuration, one for logs, and one for status. 

For the sake of simplicity, it might be best to have the initrd in the VM spawn the application and then wait for it. It would then return the exit status of the app to the host over the vsock. The only option on both zero and non-zero exit status would be to halt the VM. The policy of what to do on app exiting would then be handled on the host by northstar.

pclesr, 2020-11-11:
For logging, it may be sufficient to have the application in a VM log everything to stdout/stderr and then pipe all output to a vsock. On the host, then a corresponding thread would just dump it to stdout, which would be captured by the normal northstar log wrapping mechanisms.

For example, as a test, a shell script in a VM which just prints something to stdout once a second. The output is send to a pipe (nc-vsock) that sends it to the host. On the host, there is just a Socat listening on the pipe.

In this example the port number is 40, chosen at random
```
VM:
/ # /tmp/doit.sh | nc-vsock 2 40
Host:
socat UNIX-LISTEN:./fcvsock.sock_40 -                                                             <
count 0
count 1
count 2
etc
```

pclesr, 2020-12-10:
```
Files to bring up VM:
  vmconfig=$rundir/vmconf.json
        VM config, passed to VM by config vsock
                network config
                mount config
                application config
  fcconfig=$rundir/fcconf.json
        Firecracker config, passed to firecracker
                path to kernel, initrd
                path to additional mounts
                vtap(network) config
                machine config
                path to vsock and config
                path to metrics file
  vmstatus=$rundir/vmstatus
        Status from VM, integer
  vmlog=$rundir/vmlog
        All output from app in VM, from vsock
  fclog=$rundir/fclog
        Firecracker log output
  fcmetrics=$rundir/fcmetrics
        Firecracker metrics
  ctlsock=$rundir/fcsock
        Control socket for firecracker; may not be req.
        Can be used to force metrics dump
  vsock=$rundir/fcvsock.sock
        vsock base name; there are 3 ports used. The naming
        convention is to append the port to the name
          0  config to VM (stdin)
          1  logs (stdout)
          2  status (stderr)

Socat servers:
        socat is used as an interface to the VM. It does read/write on
        the vsock that firecracker uses as a communications channel to the
        VM.

        config server (port 0)
        Waits until base socket is created by FC (rundir/fcvsock.sock)
        then starts listening on port 0. VM will connect as client
        and retrieve json config file

        log server (port 1)
        Starts listening on port 1 and directs all output to file. Uses
        'fork' semantics so that multiple connections are allowed from
        the VM. All data received is written to a file. This is to avoid
        console output (which is slow over the serial console)

        status server (port 2)
        When the application exits in the VM, the exit code is sent
        to the status server. This is then reported as the exit
        code of the container. The exit code of firecracker itself
        can not be used because it does not reflect the actual application
        status itself.
```

flxo, 2022-05-02:
Archive fc issues

closed 2022-05-02 by flxo