esrlabs/northstar #172: Provide static network configuration in container manifest

marcmo, 2020-11-10:
This implies that the entity that is integrating/deploying containers to a target must be responsible for ensuring addresses are valid, there are no collisions, and containers/VMs are reachable only as intended
	
- IPv4 address of the container/VM
- IPv4 netmask
- IPv4 gateway
  - IPv4 address of host ?
- bridge name
- device type (tap or veth)
  - device name

 ?

pclesr, 2020-11-11:
NorthCracker requires a tap device which should be created before the VM starts, and then torn down after it exits. This tap device must be connected to a bridge that is doing masquerading of IP addresses.

Only one bridge is required, and all tap devices can connect to it. This is analogous to attaching all the `veth` devices for network namespaces to the bridge. The bridge script that was developed during initial benchmarking  provides this functionality; it is started at boot time before northstar, and creates a bridge named `nstbr1` that all VMs/containers can attach to.  I would propose that we keep this functionality for the sake of simplicity, and have all VMs/containers reference this "well-known" bridge name, analogous to all Docker containers using a bridge named `docker0`. 

I will follow up on @flxo suggestion about using `impl` in the process trait to provide setup/teardown of the required devices.

Maybe something like the following in the manifest ?
```
netconf:
  ipv4:
    addr: 192.168.200.10
    cidr: 24
    gw: 192.168.200.1
    host_addr: 192.168.0.10
  tap_dev:
    name: tap0
  veth_dev:
    name: {}
```

pclesr, 2020-11-13:
I think that if we use a network namespace for the VM, then we don't need to specify the device name (tap0), since it will be unique to the namespace. This would simplify the manifest handling. 

The downside is that it will add overhead to the launching since the namespace will need to be created; however, it might be possible to do the namespace creation asynchronously at the same time as the verity devices are set up since we will be waiting for IO to complete anyway.

pclesr, 2020-11-18:
Using network namespaces means we don't need to specify a IP configuration at all. Just like docker, the IP address can be assigned automatically from a 172.16.0.0 subnet.

The integrator will assign an index, from 0-254 for each namespace in the manifest. That is the only configuration that is necessary, and is done to ensure there are no duplicate addresses.

The index is used as a subnet, and then the IP address for the namespace and VM is assigned from that subnet. The address of the namespace and VM is identical in the final ipv4 octet in each subnet. (10 for namespace, 20 for VM)

The manifest might look like this:
```
netconf:
  namespace: true
  virt_machine: false
  index: 5
```
This would result in a container with an IP address of 172.16.5.10 .

 If the manifest specified
```
netconf:
  namespace: true
  virt_machine: true
  index: 5
```
This would result in a namespace with an IP address of 172.16.5.10 and a VM with an address of 172.16.5.20 . Note that for a VM, the network namespace is not visible to the application; it only sees and exists in the VM. Note that it is not possible to specify` virt_machine:true` and `namespace:false` 

Implementation detail:
When the north runtime starts, it creates a network bridge in the default namespace (just like docker) and assigns it an address of 172.16.0.1/16 . Being a /16 network means we can create subnets from it. The iptables are set up to masquerade all addresses from 172.16.0.0/16 .

Namespaces are created with `veth` devices, and they are always created in pairs.

When creating a VM, a namespace is created, and one end of the `veth` pipe is attached to the bridge in the default namespace. A `macvtap` device is created in the namespace attached to the other end of the `veth` in the namespace, and firecracker is given this device to use as it's network communication pipe. 

Since the macvtap is created in a namespace, the name generated by the kernel ends up being the same for all namespaces (tap2, since the `ifindex` is 2) . This means that a unique character device node must be created with the major/minor specified in the uevent (`/sys/class/net/$macvtap_dev/macvtap/tap2/uevent`) . This is the device that is handed to firecracker.

The VM is given a /16 address (172.16.5.20 in the above example) and the gateway is set to the bridge in the default namespace.  

I don't think it is necessary to require the host address in the manifest; it should be possible to reach services on the host by using the gateway address, but I have not verified this.

flxo, 2022-05-02:
Archive fc issues

closed 2022-05-02 by flxo