esrlabs/northstar #174: Configuration shall refer to a file that contains the key and not a directory

flxo, 2020-11-10:
Currently the keys are referenced in the configuration with a directory. The runtime searches for files within that directory which is not intuitive. 

The configuration should look like this:

```toml
debug = true
console_address = "localhost:4200"
key = "examples/keys/north.key"
container_uid = 1000
container_gid = 1000

[directories]
container_dirs = [ "target/north/registry" ]
run_dir = "target/north/run"
data_dir = "target/north/data"
```

bzld, 2020-11-10:
The `signature.yaml` contains a `key:` field to support having different signing keys between NPKs (owned by different parties for example).
That's why the folder approach was used to search for the right one to use.

Maybe we don't need this flexbility.

In case we _do_ want to keep it, we should rather have a key name -> key file mapping in configuration? Like:
```toml
[keys]
north = "examples/keys/north.pub"
external = "examples/keys/external.pub"
```

flxo, 2020-11-10:
Or even tie it up to the `container_dir`:

```
[registries] // Not sure if that's a good name at all
foo:
   key: /system/etc/north/blah...
bar:
```
This would also allow to drive a "registry" dir without verity.

bzld, 2020-11-10:
Yes, interesting middle ground 😉 
I think the right name for that would be `[repositories]` instead of `[registries]`?

flxo, 2020-11-10:
I tried to sketch this in `config.rs`:

* The key is optional for a repository and indicated that the npk shouldn't be mounted via verity
* `runtime::state::State::applications` gets a reference to a `Repository`
* `runtime::state::State::signing_keys` is changed and holds a key for each repository.
* The `writeable` flag indicated whether npk can be added/removed at runtime
* The `install`/`uninstall` api commands must get a repository id
* Npks with the same name in the manifest are not allowed
* `api::Container` gets a repository id
* `api::Request::Repositories` is added and returns a `Map<api::RepositoryId, api::Repository>` (note the `api::`)
 

```rust
/// Identification of a repository. This is used during add and remove operations
/// to identify the location.
type RepositoryId = String;

#[derive(Clone, Debug, Deserialize)]
pub struct Repository {
    /// Repository directory
    pub dir: PathBuf,
    /// Writeable
    pub writeable: bool,
    /// Signing key. If this is None the containers are mounted without verity
    pub key: Option<PathBuf>,
}

#[derive(Clone, Debug, Deserialize)]
pub struct Config {
    /// Print debug logs
    pub debug: bool,
    /// Console address
    pub console_address: String,
    /// Container UID
    pub container_uid: u32,
    /// Container GID
    pub container_gid: u32,
    /// Directory where containers are mounted at runtime. This must
    /// be read and writeable by the runtime.
    pub run_dir: PathBuf,
    /// Directory where rw data of container shall be stored. Thius
    /// must be read and writeable by the runtime.
    pub data_dir: PathBuf,
    /// List of repositories
    pub repositories: HashMap<RepositoryId, Repository>,
    /// CGroup configuration
    pub cgroups: CGroups,
    /// Platform specific device configuration
    pub devices: Devices,
}

@bzld Thoughts?

bzld, 2020-11-13:
@flxo Looks pretty good.

Small point: for (future?) flexibility, the manifests contain a key name besides the actual signature. In your approach key names don't exist and there would be just the one key set for the repository. So we would just ignore the key name and try to check if the signature works or not? Then we might consider to remove the key name field from the manifest even.

flxo, 2020-11-13:
@bzld Agree. I don't think that we need the complexity of multiple keys per Repository. This could also be easily added later if needed.

closed 2021-01-13 by alfonsoros88