esrlabs/northstar #181: create and teardown default network bridge at startup/exit

pclesr, 2020-11-12:
In the POC, a script was run at boot time to create a network bridge that did masquerading. This bridge is used for both network namespaces (veth devices) and VMs (tap devices). The script` netns_br.sh` in the old repo did this; the equivalent functionality must be brought into the north runtime. 

The toml file should have a new switch that enables this functionality.

At startup:
-   Create bridge using `ip` utility
-   User must specify the IP address (and optionally MAC address)  in the toml file
--  Assign it a MAC address (use 00A0D9 CONVEX COMPUTER CORPORATION) by default, since that is guaranteed to be available
-   Set up masquerading using `iptables`
-   Enable ip forwarding

Teardown is the reverse

pclesr, 2020-11-12:
Maybe something along the following in the toml file:

```
[bridge]
enabled = true
br_name = "nstar0"
br_ipv4 = "192.168.2.10"
br_ipv4_cidr = 24
br_mac = "00.A0.D9.CA.FE.DD"
```

pclesr, 2020-11-18:
After working with namespaces to get VMs to work, I think it can be much simpler. I would propose that we do not allow bridge configuration for now to make it easier. Just as with Docker, if the bridge is enabled (which is required for network namespaces), it will create a default bridge with a default IP address.

The ipaddress must be a `/16` , like 172.16.x.x . It could also be a 192.168 or a 10.x . The only reason to allow a custom address range would be to avoid collisions with another container runtime (like docker).

So, the configuration in `north.toml `would be 
`bridge: true
`
This would result in a bridge named `nstbr1` being created with an address of `172,.16.0.1/16 `and a MAC address of `00.a0.d9.ca.fe.dd` . It would also set up `iptables` to do masquerading on the` /16` address range.

Whenever the north runtime starts, it would need to check if the bridge exists; if not it must be created. During graceful shutdown, the bridge should be torn down and the `iptable` entries removed.

pclesr, 2020-11-26:
The north.toml file will have the following:
```
[bridge]
enabled = true
ipv4_slash16 = "172.30.0.0"
```
If enabled is true, then the class B portion of the IP address (172.30 in this example) will be used to configure the bridge. If set to false, then the address will be set to the empty string. Network configuration is controlled by the container manifest. The manifest now has an optional section :
```
netconf:
  enable_namespace: true
  namespace_uniq_subnet: 42
```
If `enable_namespace` is true, then it will attempt to configure the bridge using the class B portion of the IP address specified in the `north.toml` file upon first use.

Stated differently, a bridge is lazily configured. It is not configured until the first container with `enable_namespace:true` is launched. If the bridge is already configured, no configuration happens, and the existing bridge is used.

The defaults for the name and macaddr are hard-coded, since there is no need for them to be configurable.

The teardown of the bridge will happen during graceful shutdown. If north crashes, the bridge is left in place

flxo, 2022-05-02:
Archive fc issues

closed 2022-05-02 by flxo