esrlabs/northstar #189: Integration tests

flxo, 2020-11-13:
The integration test currently do happy path testing of the examples. Extend it with:

* API Tests
* Stress tests: eg. start/stop in a loop
* Signatures
* Malformed manifests
* Security features
** Signal on seccomp violation
* CGroups
** Verify cgroup settings
** Test cgroup oom
* ...

pclesr, 2020-12-04:
I would also add:

malformed north.toml
start/stop runtime in a loop (not just a container)

network settings:
I have a container that I am using to verify network namespace functionality that I will check in as part of the network namespace PR. This container can be used as a basis and converted/integrated into the test framework, but it will require running on the real target (I have not tried it on an x86 dev host)

flxo, 2021-05-19:
Split into #339 and #340

closed 2021-05-19 by flxo