esrlabs/northstar #194: Android: expose /dev/sockets to applications

flxo, 2020-11-13:
Minijail exposes only a subset of `/dev` to the process. On Android at least /dev/socket/logdw` is needed to reach `logd`. This can be accomplished by stupidly bind mounting `/dev` but this is less restrictive than the MJ way.

Another option is to bind mount `/dev/socket` to `/dev/socket`. Or symlink(?) dedicated devices from `/dev` on a manifest configuration basis.

flxo, 2020-11-18:
The minijail `mount_dev` shall be optional. Therefore the `manifest::Mount` get's a new entry

```rust
    Dev(Dev)
    }

enum Dev {
   Minimal,
   Full,
   Custom(Vec<PathBuf>), // Not sure if this is feasible
}
```

flxo, 2020-11-18:
Maybe the Minimal is a implicit default.

closed 2020-11-23 by flxo

bzld, 2020-12-01:
Another interesting option: Linux supports bind-mounting **files** too (but you must `touch` the target file first)
That seems to work for unix domain sockets as well, as long as the server is already bound/listening when bind-mounting:

Shell1:
```
mkdir test1
nc -v -U -l test1/socket
```

Shell2:
```
mkdir test2
touch test2/socket
sudo mount -o bind test1/socket test2/socket
nc -v -U test2/socket
```

PR #203 (2020-11-23): Add /dev mount options to the manifest
Fixes #194