esrlabs/northstar #212: Debugging containers is hard

flxo, 2020-11-25:
Currently debugging containers is a hard thing. The mount namespace prevents the user to have a look inside etc....

Ideas:

* Provide a `strace` mode where all system call of a container get traced
* Provide a way to access `stdin` of a container in order to run something interactive e.g a shell
* ...

pclesr, 2020-12-18:
As a very first step, it would be helpful if it were possible to just mount the image on the host, along with the bind mounts specified in the manifest, plus the system root. Then, spawn a shell in the chroot'd environment.

In this way, it would be similar to a docker exec of a shell. The main goal would to check permissions, paths, etc, and to verify the steps in launching the target application.

flxo, 2021-05-19:
Today we have strace that can be opt in for a container launch. The example main provides a flag to disable the mount namespace for rootfs inspection. Checkin the rootfs is also easily possible via `/proc/<pid>/root`.

Getting a shell is not so easy since there no `stdin` and the shell must be present in the container.

flxo, 2022-09-02:
Closing this because `strace` is considered the the most useful thing in place along with logs.

closed 2022-09-02 by flxo