esrlabs/northstar #266: Need tests to exercise process cleanup code

pclesr, 2021-01-28:
We need a test that ensures that all child processes of a container are cleaned up.

For example, if a container spawns 3 processes, upon exit (or nstar stop), we must make sure that all processes are killed.

flxo, 2021-05-19:
TODO: Create an integration test  that uses `ps` pre and post some start/stop of the runtime and containers. This test should also `SIGKILL` the runtime.

The problem today is that we use [PR_SET_PDEATHSIG](https://man7.org/linux/man-pages/man2/prctl.2.html) in the child processes. But unfortunately this is *not* done by the minijail and island init. The background is that the `clone` happens from a (possibly) short lived thread of the tokio runtime that may be terminated but the thread pool implementation after some timeout. `PR_SET_PDEATHSIG` sends the configured signal upon the termination of the *thread* that cloned and not the process.

flxo, 2021-06-09:
@inorick As we talked about this yesterday - you may want to have a try

inorick, 2021-06-23:
Init processes are now cleaned up if the runtime dies (https://github.com/esrlabs/northstar/pull/370)

closed 2021-06-23 by inorick