esrlabs/northstar #288: Need long running tests to create churn

pclesr, 2021-02-04:
We need a long running test that starts and stops hundreds of containers over a period of time to generate churn. This is extremely helpful in checking for resource leaks.

flxo, 2021-05-19:
@alfonsoros88 created some scripts that can be used on a self hosted runner. The  https://github.com/esrlabs/northstar/tree/master/tools/stress utility is something similar. 
GH action jobs can run for up to 6h in the GH environment without a self hosted runner and all it's implications.

Maybe we can dedicated workflow for this? Trigger conditions could be e.g pushes to a dedicated branch.

flxo, 2022-04-07:
Long running tests are in place but not public. If interested in the setup, feel free to ping here.

closed 2022-04-07 by flxo