esrlabs/northstar #316: Runtime error: Key: Invalid key file

PatrickKohl, 2021-03-15:
Hi, I got some Error after starting northstar:

**2021-03-15 04:57:12.934 29836 2754725584 I northstar: Northstar v0.6.2 (619fcc2)
2021-03-15 04:57:12.934 29836 2754725584 D northstar: Entering mount namespace
2021-03-15 04:57:12.935 29836 2754725584 D northstar: Creating target/northstar/data
2021-03-15 04:57:12.935 29836 2754725584 D northstar: Creating target/northstar/run
2021-03-15 04:57:12.935 29836 2754725584 D northstar: Creating log
2021-03-15 04:57:12.935 29836 2745528240 D northstar: Loading key examples/keys/northstar.pub
2021-03-15 04:57:12.935 29836 2745528240 E northstar: Runtime error: Key: Invalid key file: examples/keys/northstar.pub not a file or has '.pub' extension
2021-03-15 04:57:12.935 29836 2754725584 W northstar: Runtime exited with Key(KeyFile("examples/keys/northstar.pub not a file or has \'.pub\' extension"))**

What can I do?

flxo, 2021-03-15:
Hello @PatrickKohl 

You have a relative path to the key in your configuration (either from a file or constructed in application - depends how you start northstar).
You probably started northstar from  different directory that the northstar workspace root where the example key is located in the relative location logged in the trace.

You can check your `pwd`.

PatrickKohl, 2021-03-15:
works, thanks

PatrickKohl, 2021-03-15:
Ok, but only until:
➜ start hello-world 0.0.1
✗ failed: Mount("Device mapper error: IoCtrl(Sys(EINVAL))")

flxo, 2021-03-15:
@PatrickKohl It's hard to help without full traces config etc. Maybe you can provide such information.

My guess is that your kernel doesn't support verity or you're running without sufficient permissions.

PatrickKohl, 2021-03-15:
Yes, sorry.

**~/northstar$ sudo ./target/release/northstar
2021-03-15 13:48:48.789 23604 2176079568 I northstar: Northstar v0.6.2 (619fcc2)
2021-03-15 13:48:48.789 23604 2176079568 D northstar: Entering mount namespace
2021-03-15 13:48:48.791 23604 2176079568 D northstar: Creating target/northstar/data
2021-03-15 13:48:48.791 23604 2176079568 D northstar: Creating target/northstar/run
2021-03-15 13:48:48.791 23604 2176079568 D northstar: Creating target/northstar/logs
2021-03-15 13:48:48.791 23604 2166882224 D northstar: Loading key examples/keys/northstar.pub
2021-03-15 13:48:48.847 23604 2168983472 D northstar: Starting console on 127.0.0.1:4200
2021-03-15 13:48:48.847 23604 2168983472 D northstar: Started console on 127.0.0.1:4200
2021-03-15 13:49:03.090 23604 2173185968 D northstar: Client "127.0.0.1:48202" connected
2021-03-15 13:51:31.923 23604 2171084720 I northstar: Trying to start hello-world:0.0.1
2021-03-15 13:51:31.924 23604 2171084720 I northstar: Need to mount 1 container before starting hello-world:0.0.1
2021-03-15 13:51:31.936 23604 2173185968 D northstar: Mounting hello-world:0.0.1
2021-03-15 13:51:31.936 23604 2173185968 D northstar: Loaded manifest of hello-world:0.0.1
2021-03-15 13:51:31.936 23604 2173185968 D northstar: Acquired control lock
2021-03-15 13:51:31.937 23604 2173185968 D northstar: Using loop dev /dev/loop1
2021-03-15 13:51:31.937 23604 2173185968 D northstar: Setting offset 4096 and limit 909312
2021-03-15 13:51:31.953 23604 2173185968 D northstar: Released control lock
2021-03-15 13:51:31.976 23604 2173185968 D northstar: Loopback setup took 0.040s
2021-03-15 13:51:31.977 23604 2173185968 D northstar: Loop device id is 7:1
2021-03-15 13:51:31.977 23604 2173185968 D northstar: Creating a read-only verity device (name: northstar_23604_hello-world_0.0.1)
2021-03-15 13:51:31.977 23604 2173185968 D northstar: Creating verity device
2021-03-15 13:51:31.981 23604 2173185968 D northstar: Using verity device /dev/dm-0
2021-03-15 13:51:32.257 23604 2171084720 D northstar: Failed to mount hello-world:0.0.1: Mount: Device mapper error: IoCtrl(Sys(EINVAL))
2021-03-15 13:51:32.257 23604 2171084720 W northstar: Failed to start hello-world:0.0.1: Mount: Device mapper error: IoCtrl(Sys(EINVAL))
2021-03-15 13:52:01.397 23604 2164780976 I northstar: Trying to start ferris:0.0.1
2021-03-15 13:52:01.398 23604 2164780976 I northstar: Need to mount 1 container before starting ferris:0.0.1
2021-03-15 13:52:01.408 23604 2166882224 D northstar: Mounting ferris:0.0.1
2021-03-15 13:52:01.408 23604 2166882224 D northstar: Loaded manifest of ferris:0.0.1
2021-03-15 13:52:01.409 23604 2166882224 D northstar: Acquired control lock
2021-03-15 13:52:01.409 23604 2166882224 D northstar: Using loop dev /dev/loop1
2021-03-15 13:52:01.410 23604 2166882224 D northstar: Setting offset 4096 and limit 942080
2021-03-15 13:52:01.425 23604 2166882224 D northstar: Released control lock
2021-03-15 13:52:01.440 23604 2166882224 D northstar: Loopback setup took 0.032s
2021-03-15 13:52:01.441 23604 2166882224 D northstar: Loop device id is 7:1
2021-03-15 13:52:01.441 23604 2166882224 D northstar: Creating a read-only verity device (name: northstar_23604_ferris_0.0.1)
2021-03-15 13:52:01.441 23604 2166882224 D northstar: Creating verity device
2021-03-15 13:52:01.446 23604 2166882224 D northstar: Using verity device /dev/dm-1
2021-03-15 13:52:01.485 23604 2164780976 D northstar: Failed to mount ferris:0.0.1: Mount: Device mapper error: IoCtrl(Sys(EINVAL))
2021-03-15 13:52:01.485 23604 2164780976 W northstar: Failed to start ferris:0.0.1: Mount: Device mapper error: IoCtrl(Sys(EINVAL))
2021-03-15 13:52:26.529 23604 2171084720 I northstar: Trying to start crashing:0.0.1
2021-03-15 13:52:26.530 23604 2171084720 I northstar: Need to mount 1 container before starting crashing:0.0.1
2021-03-15 13:52:26.540 23604 2159669168 D northstar: Mounting crashing:0.0.1
2021-03-15 13:52:26.541 23604 2159669168 D northstar: Loaded manifest of crashing:0.0.1
2021-03-15 13:52:26.542 23604 2159669168 D northstar: Acquired control lock
2021-03-15 13:52:26.542 23604 2159669168 D northstar: Using loop dev /dev/loop1
2021-03-15 13:52:26.544 23604 2159669168 D northstar: Setting offset 4096 and limit 909312
2021-03-15 13:52:26.557 23604 2159669168 D northstar: Released control lock
2021-03-15 13:52:26.572 23604 2159669168 D northstar: Loopback setup took 0.031s
2021-03-15 13:52:26.573 23604 2159669168 D northstar: Loop device id is 7:1
2021-03-15 13:52:26.573 23604 2159669168 D northstar: Creating a read-only verity device (name: northstar_23604_crashing_0.0.1)
2021-03-15 13:52:26.573 23604 2159669168 D northstar: Creating verity device
2021-03-15 13:52:26.576 23604 2159669168 D northstar: Using verity device /dev/dm-2
2021-03-15 13:52:26.609 23604 2171084720 D northstar: Failed to mount crashing:0.0.1: Mount: Device mapper error: IoCtrl(Sys(EINVAL))
2021-03-15 13:52:26.609 23604 2171084720 W northstar: Failed to start crashing:0.0.1: Mount: Device mapper error: IoCtrl(Sys(EINVAL))**

log_level = "DEBUG"
console = "tcp://localhost:4200"
run_dir = "target/northstar/run"
data_dir = "target/northstar/data"
log_dir = "log"

[repositories.default]
dir = "target/northstar/repository"
writable = true
key = "/home/boerr/northstar/target/release/northstar.pub"

[cgroups]
memory = "northstar"
cpu = "northstar"

**[devices]
unshare_root = "/"
loop_control = "/dev/loop-control"
loop_dev = "/dev/loop"
device_mapper = "/dev/mapper/control"
device_mapper_dev = "/dev/dm-"**

flxo, 2021-03-15:
Did you check that your kernel supports `squashfs` and `dm-verity`? I expect that you're running on a desktop like system where normally at least the squashfs support is available in a loadable module - but not loaded by default.

PatrickKohl, 2021-03-18:
Its a vanilla nvidia Jetson Nano. When I run script : check_conf.sh

bash ./check_conf.sh
dm verity module appears to be missing
Check CONFIG_DM_VERITY. Also enable CONFIG_DM_VERITY_FEC
The following are also required:
    CONFIG_DM_CRYPT
    CONFIG_DM_THIN_PROVISIONING
    CONFIG_CRYPTO_SHA256

At Xeon with Ubuntu 20.04 Server the demos run like described.

flxo, 2021-03-19:
If your board doesn't support verity you can remove the key from the repository config section (it's optional) to mount the images without verity. Otherwise get a kernel with verity support ;-)

flxo, 2021-03-26:
@PatrickKohl ? Any news? Otherwise I'd close this.

PatrickKohl, 2021-03-26:
It worked, thank you for good documentation, can be closed.

PatrickKohl, 2021-04-07:
Hi, I recompiled Kernel with Verity target support, but got an error again:
[image]
Can it be that need a spezial squashfs in the Kernel?

flxo, 2021-04-08:
There's nothing special about squashfs in the kernel. The only thing that needs to be assured is that your squashfs driver supports the same compression algorithm that you used when building the npks. This defaults to `gzip` but can be overwritten on the CLI.

PatrickKohl, 2021-04-08:
It was, because of "Tegra SE driver for crypto algorithms" in Kernel. After disabling the feature it is working.

closed 2021-04-08 by PatrickKohl