esrlabs/northstar #318: help command is not helpful

pclesr, 2021-03-16:
The help command for the start command is missing
```
help
    containers      List available containers
    help            Print help
    install         Install a container
    mount           Start a container
    quit            Quit
    repositories    List configured repositories
    shutdown        Shutdown Northstar
    start
    stop            Stop a container
    umount
```

It does not indicate that one must now also add the version number to the `start` command.

pclesr, 2021-03-16:
Also, I don't know where the gifs are coming from, but they should be removed as they may not render correctly on all terminals.
```
 uninstall       Uninstall a container✔ success
📣  Started(Container { inner: Inner { name: "hello-world", version: Version { major: 0, minor: 0, patch: 1, pre: [], build: [] } } })
➜ stop hello-world 0.0.1
📣  Stopped(Container { inner: Inner { name: "hello-world", version: Version { major: 0, minor: 0, patch: 1, pre: [], build: [] } } })
```

alfonsoros88, 2021-03-16:
I'll try to address these issues in #299

commit 378c8fe4 (2021-03-26): Validate input with clap

closes #318

Also removes all the emojis in nstar
closed 2021-03-26 by flxo