esrlabs/northstar #319: Detailed documentation about building a northstar package

PatrickKohl, 2021-03-22:
Hi, are there some detailed documentation out there to build a package?

I got with:

 **./target/release/sextant pack --manifest ./echo/manifest.yaml --out ./echo/  --root ./echo/**

this:

**✗ failed: Npk("Archive error: Failed to locate signature.yaml in ZIP file")**

flxo, 2021-03-22:
@PatrickKohl You didn't specify a key when packing the npk. This is totally valid. Next you placed the npk in a repository that has a key configured. Your npk doesn't get mounted because it's not signed and doesn't contain the signatures.

The error log could give a (better) hint about that. @inorick Can you please add this?

PatrickKohl, 2021-03-22:
Sorry again:

**2021-03-21 21:10:07.134 3849052 640530816 I northstar: Northstar v0.6.2 (a38715a)
2021-03-21 21:10:07.134 3849052 640530816 D northstar: Entering mount namespace
2021-03-21 21:10:07.134 3849052 640530816 D northstar: Creating target/northstar/data
2021-03-21 21:10:07.134 3849052 640530816 D northstar: Creating target/northstar/run
2021-03-21 21:10:07.134 3849052 640530816 D northstar: Creating target/northstar/logs
2021-03-21 21:10:07.134 3849052 636323584 D northstar: Loading key examples/keys/northstar.pub
2021-03-21 21:10:07.140 3849052 632108800 D northstar: Starting console on 127.0.0.1:4200
2021-03-21 21:10:07.140 3849052 632108800 D northstar: Started console on 127.0.0.1:4200
2021-03-21 21:10:08.684 3849052 640526080 D northstar: Client "127.0.0.1:45460" connected
2021-03-21 21:10:21.137 3849052 634222336 I northstar: Trying to start echo:0.0.1
2021-03-21 21:10:21.137 3849052 634222336 I northstar: Need to mount 1 container before starting echo:0.0.1
2021-03-21 21:10:21.137 3849052 634222336 W northstar: Failed to start echo:0.0.1: NPK error: Zip { context: "Failed to locate signature.yaml in ZIP file", error: FileNotFound }
2021-03-22 07:49:39.948 3849052 640526080 I northstar: 127.0.0.1:45460: Connection closed
2021-03-22 07:49:39.948 3849052 640526080 I northstar: 127.0.0.1:45460: Connection closed**

inorick, 2021-03-23:
There is a new guide on working with NPKs here:
https://esrlabs.github.io/northstar/sextant/npk.html

Let me know if you find any issues with that.

flxo, 2021-03-26:
Reopen if needed.

closed 2021-03-26 by flxo