esrlabs/northstar #334: Permission denied when using GPU in a container

yrjyrj123, 2021-04-30:
Dear all develops,

I try to use Northstar on Jetson Nano with GPU, but it doesn't work.

I run **strace** in a container, and find the process can't get the permission to access device file in **/dev**.

[image]

I have built a kernel which can pass the test in **check_conf.sh**, and I can run the demo containers.

In the **manifest.yml**, I add "/dev: full" in the **mounts** section.

And I can see /dev in container with "ls -al /dev".

[image]

Can you help me think about the possible reasons?

Thanks.

flxo, 2021-04-30:
Hello @yrjyrj123 

Can you please post the manifest? Did you ensure that the uid/gid you configured has permission to access `/dev/nvhost-ctrl-gpu`?

yrjyrj123, 2021-05-11:
Thanks for your response. 😊

This is my manifest.yaml:

> name: cuda_demo
> version: 0.0.2
> init: "/deviceQuery"
> args: []
> uid: 1000
> gid: 1000
> 
> mounts:
>   /dev: full
>   /lib:
>     host: /lib
>   /usr:
>     host: /usr 
>   /var:
>     host: /var  
>   /opt:
>     host: /opt
>   /etc:
>     host: /etc 
>   /boot:
>     host: /boot
>   /sys:
>     host: /sys
>   /bin:
>     host: /bin  
> 
> io:
>   stdout: pipe
>   stderr: pipe

This is the log when I try to run a container.

> 2021-05-11T03:34:50.638Z: DEBUG: runtime::console Client 127.0.0.1:36886 connected
> 2021-05-11T03:34:50.638Z: INFO : runtime::state Trying to start cuda_demo:0.0.2
> 2021-05-11T03:34:50.638Z: INFO : runtime::state Need to mount 1 container before starting cuda_demo:0.0.2
> 2021-05-11T03:34:50.640Z: DEBUG: runtime::mount Mounting cuda_demo:0.0.2
> 2021-05-11T03:34:50.640Z: DEBUG: runtime::mount Loaded manifest of cuda_demo:0.0.2
> 2021-05-11T03:34:50.640Z: DEBUG: runtime::loopdev Acquired control lock
> 2021-05-11T03:34:50.640Z: DEBUG: runtime::loopdev Using loop dev /dev/loop1
> 2021-05-11T03:34:50.641Z: DEBUG: runtime::loopdev Setting offset 4096 and limit 184320
> 2021-05-11T03:34:50.670Z: DEBUG: runtime::loopdev Released control lock
> 2021-05-11T03:34:50.698Z: DEBUG: runtime::loopdev Loopback setup took 0.058s
> 2021-05-11T03:34:50.698Z: DEBUG: runtime::mount Loop device id is 7:1
> 2021-05-11T03:34:50.698Z: DEBUG: runtime::mount Creating a read-only verity device (name: northstar_16474_cuda_demo_0.0.2)
> 2021-05-11T03:34:50.699Z: DEBUG: runtime::mount Creating verity device
> 2021-05-11T03:34:50.704Z: DEBUG: runtime::mount Using verity device /dev/dm-0
> 2021-05-11T03:34:50.707Z: DEBUG: runtime::mount Resuming device
> 2021-05-11T03:34:50.707Z: DEBUG: runtime::mount Waiting for device /dev/dm-0
> 2021-05-11T03:34:50.707Z: DEBUG: runtime::mount Verity setup took 0.009s
> 2021-05-11T03:34:50.707Z: DEBUG: runtime::mount Creating mount point target/northstar/run/cuda_demo:0.0.2
> 2021-05-11T03:34:50.708Z: DEBUG: runtime::mount Mounting squashfs fs on /dev/dm-0 to target/northstar/run/cuda_demo:0.0.2
> 2021-05-11T03:34:50.711Z: DEBUG: runtime::mount Mounting took 0.003s
> 2021-05-11T03:34:50.711Z: INFO : runtime::mount Mounted cuda_demo:0.0.2 Mounting: 0.071s
> 2021-05-11T03:34:50.711Z: DEBUG: runtime::state Successfully mounted cuda_demo:0.0.2
> 2021-05-11T03:34:50.712Z: INFO : runtime::state Creating cuda_demo:0.0.2
> 2021-05-11T03:34:50.712Z: DEBUG: runtime::minijail Setting UID to 1000
> 2021-05-11T03:34:50.712Z: DEBUG: runtime::minijail Setting GID to 1000
> 2021-05-11T03:34:50.712Z: DEBUG: runtime::minijail Mounting /bin on /bin
> 2021-05-11T03:34:50.712Z: DEBUG: runtime::minijail Mounting /sys on /sys
> 2021-05-11T03:34:50.712Z: DEBUG: runtime::minijail Mounting /etc on /etc
> 2021-05-11T03:34:50.712Z: DEBUG: runtime::minijail Mounting /lib on /lib
> 2021-05-11T03:34:50.712Z: DEBUG: runtime::minijail Mounting /usr on /usr
> 2021-05-11T03:34:50.712Z: DEBUG: runtime::minijail Mounting /boot on /boot
> 2021-05-11T03:34:50.712Z: DEBUG: runtime::minijail Mounting /opt on /opt
> 2021-05-11T03:34:50.712Z: DEBUG: runtime::minijail Mounting /var on /var
> 2021-05-11T03:34:50.712Z: DEBUG: runtime::minijail Preparing "/deviceQuery"
[... 47 lines cut ...]
> 2021-05-11T03:37:41.404Z: DEBUG: runtime::console Client 127.0.0.1:36892 connected
> 2021-05-11T03:37:41.404Z: INFO : runtime::state Trying to start cuda_demo:0.0.2
> 2021-05-11T03:37:41.404Z: INFO : runtime::state Need to mount 1 container before starting cuda_demo:0.0.2
> 2021-05-11T03:37:41.406Z: DEBUG: runtime::mount Mounting cuda_demo:0.0.2
> 2021-05-11T03:37:41.406Z: DEBUG: runtime::mount Loaded manifest of cuda_demo:0.0.2
> 2021-05-11T03:37:41.407Z: DEBUG: runtime::loopdev Acquired control lock
> 2021-05-11T03:37:41.407Z: DEBUG: runtime::loopdev Using loop dev /dev/loop1
> 2021-05-11T03:37:41.410Z: DEBUG: runtime::loopdev Setting offset 4096 and limit 184320
> 2021-05-11T03:37:41.428Z: DEBUG: runtime::loopdev Released control lock
> 2021-05-11T03:37:41.450Z: DEBUG: runtime::loopdev Loopback setup took 0.044s
> 2021-05-11T03:37:41.450Z: DEBUG: runtime::mount Loop device id is 7:1
> 2021-05-11T03:37:41.451Z: DEBUG: runtime::mount Creating a read-only verity device (name: northstar_16575_cuda_demo_0.0.2)
> 2021-05-11T03:37:41.451Z: DEBUG: runtime::mount Creating verity device
> 2021-05-11T03:37:41.453Z: DEBUG: runtime::mount Using verity device /dev/dm-0
> 2021-05-11T03:37:41.461Z: DEBUG: runtime::mount Resuming device
> 2021-05-11T03:37:41.461Z: DEBUG: runtime::mount Waiting for device /dev/dm-0
> 2021-05-11T03:37:41.461Z: DEBUG: runtime::mount Verity setup took 0.011s
> 2021-05-11T03:37:41.461Z: DEBUG: runtime::mount Creating mount point target/northstar/run/cuda_demo:0.0.2
> 2021-05-11T03:37:41.462Z: DEBUG: runtime::mount Mounting squashfs fs on /dev/dm-0 to target/northstar/run/cuda_demo:0.0.2
> 2021-05-11T03:37:41.464Z: DEBUG: runtime::mount Mounting took 0.002s
> 2021-05-11T03:37:41.464Z: INFO : runtime::mount Mounted cuda_demo:0.0.2 Mounting: 0.059s
> 2021-05-11T03:37:41.464Z: DEBUG: runtime::state Successfully mounted cuda_demo:0.0.2
> 2021-05-11T03:37:41.464Z: INFO : runtime::state Creating cuda_demo:0.0.2
> 2021-05-11T03:37:41.464Z: DEBUG: runtime::minijail Setting UID to 0
> Aborted

yrjyrj123, 2021-05-12:
I think I find the problem, it works now.

Thanks a lot.

closed 2021-05-12 by yrjyrj123