esrlabs/northstar #342: Island: Avoid allocation in seccomp

flxo, 2021-05-19:
The capability dropping in the island impl allocates and frees. This is problematic because when the rt clones there might be a lock somewhere in the libc held that causes a deadlock on *any* malloc/free in the cloned child process (in this case `init`).

Therefore after the `clone` of the `init` there should be *no* allocation. The seccomp code does in 

https://github.com/esrlabs/northstar/blob/fb430ec51dd34bd271bcae60a4d3e0f75ccfc77b/northstar/src/runtime/island/seccomp.rs#L140

and 

https://github.com/esrlabs/northstar/blob/fb430ec51dd34bd271bcae60a4d3e0f75ccfc77b/northstar/src/runtime/island/seccomp.rs#L116

Move the allocation to `clone` calling context. See #341 which is the same issue.

inorick, 2021-05-25:
BPF programs are limited to BPF_MAXINSNS (4096) instructions. We could use that limit to pre-allocate the allowlist.

flxo, 2021-05-25:
Interesting. Do we somehow check not the exceed the 4k limit? Probably 4k is so big that will never be reached ;-)

Assembling the BPF is not an expensive operation and not worth running in the child. I think that we can just generate it and pass it to `init` as we do for e.g the mounts.

inorick, 2021-05-25:
> Interesting. Do we somehow check not the exceed the 4k limit? Probably 4k is so big that will never be reached ;-)

I do not think we do but I would expect `prctl` to fail if we exceed that limit (the `seccomp` syscall does) but it is better if we do that check ourselves.

> Assembling the BPF is not an expensive operation and not worth running in the child. I think that we can just generate it and pass it to init as we do for e.g the mounts.

Agreed.

closed 2021-06-08 by flxo