esrlabs/northstar #344: Seccomp: Default set of whitelisted systemcalls for `aarch64` and define dedicated sets for glibc, musl and bionic

flxo, 2021-05-19:
Northstar performs a (libc)`execve` which internally calls tons of sys calls. Those sys calls need to be whitelisted by the seccomp configuration. The list of those sys calls is probably very c library implementation specific and might differ between e.g bionic, glibc and musl. 

https://github.com/esrlabs/northstar/blob/fb430ec51dd34bd271bcae60a4d3e0f75ccfc77b/northstar/src/runtime/island/seccomp.rs#L38-L40

flxo, 2021-05-20:
@inorick 
Actually the whitelist thought is completly wrong.
The only sys call that needs to be allowed is `execve`. Everything else is specific on the application that is being executed. Everything we've seen so far then comes from the loader etc...but again that's up to the application.

inorick, 2021-05-21:
True, the current whitelist is the sum of `execve` and all the syscall it takes to start a container with an empty `main` function. So we currenlty do account for the syscalls required by the application loader in our list.

The idea is that an empty application (empty `main` function) with an empty seccomp profile in its manifest should run successfully. At least that is what an application developer could expect.

flxo, 2021-05-21:
> The idea is that an empty application (empty `main` function) with an empty seccomp profile in its manifest should run successfully. At least that is what an application developer could expect.

But isn't that totally tied to the type of binary? We currently have the defaults for a Rust main but others...

inorick, 2021-06-01:
I cleaned out the seccomp container's `main.rs` down to an empty main function and ran it with strace twice:
1) I executed the binary extracted from the NPK without northstar ([strace_empty_container_without_northstar.txt](https://github.com/esrlabs/northstar/files/6575271/strace_empty_container_without_northstar.txt))
2) I started the NPK from the northstar runtime ([strace_empty_container_run_in_northstar.txt](https://github.com/esrlabs/northstar/files/6575275/strace_empty_container_run_in_northstar.txt))

From what I can see, all syscalls right after
```
[pid 49742] prctl(PR_SET_SECCOMP, SECCOMP_MODE_FILTER, {len=59, filter=0x7fdfc0008d30}) = 0
[pid 49742] execve("/seccomp", ["/seccomp"], 0x7fdfc0009e60 /* 2 vars */) = 0
```
also happen when you run the binary without northstar at all. This means that the only syscall we have to allow is `execve` and the rest is binary specific. This is what #353 does.

Of course, it would not hurt to verify this on other platforms.
Additionally, we could add syscalls that we consider generally safe to northstar's whitelist to make the life of container creators easier.

PR #353 (2021-06-02): Reduce REQUIRED_SYSCALLS to execve
This commit assumes that the runtime only requires execve to start containers and moves all other syscalls from the general seccomp allowlist back to the seccompt container manifest.

Fixes #344
closed 2021-06-02 by flxo