esrlabs/northstar #358: Mounting: squashfs root ownership can make mounts fail

flxo, 2021-05-31:
When mounting a squashfs image the permissions on the root directory are applied again after the mount. Creating a squshfs image from a dir with `700` and mount it - the root of the mounted fs has permissions `700` as well. This clashes with the runtime that tries to mount e.g `proc` or `tmpfs` *inside* the squashfs. Means that the runtime must run with permissions to access the directories inside the squashfs tree. A workaround is to set the permissions on the squashfs root to 777 or make it owned by the user the runtime started on.

since1q80, 2023-01-24:
Hi,
I am not quite sure whether it is the right place to put the following issue,but i am stucked in this step when i would like to start the northstar according to the Readme.
ubuntu@primary:~/Dev/northstar$ cargo run --bin northstar
    Finished dev [unoptimized + debuginfo] target(s) in 0.20s
     Running `target/debug/northstar`
2023-01-24T04:22:24.026                    northstar ⬤  INFO : Northstar Runtime v0.7.1-pre
2023-01-24T04:22:24.027                    northstar ⬤  DEBUG: Running as user ubuntu (uid: unknown)
2023-01-24T04:22:24.046                    northstar ⬤  INFO : Entering mount namespace
Error: EPERM: Operation not permitted
Anyone could help?
Thanks.

flxo, 2023-01-24:
@since1q80 

Think you're facing something else than described in this ticket. Would you mind to open a new one?

Northstar requires some permissions above normal user level. You can either run as `root` (not recommended) or grant a set of [capabilities](https://man7.org/linux/man-pages/man7/capabilities.7.html). Check [the cargo runner](https://github.com/esrlabs/northstar/blob/main/.cargo/runner.x86_64-unknown-linux-gnu) which capabilities are needed.

You probably also get happy by running Northstar with 

`cargo run` 

instead of calling the binary directly.

cheers,

@flxo

since1q80, 2023-01-24:
@flxo thanks a lot.