esrlabs/northstar #371: Rust 1.53.0 warns about undefined behaviour in bindgen generated test

flxo, 2021-06-23:
Run `argo test --features rt-island --no-default-features`:

```
...
warning: dereferencing a null pointer
   --> /home/felix/northstar/target/debug/build/northstar-789102d2595eaf1c/out/seccomp_bindings.rs:173:19
    |
173 |         unsafe { &(*(::std::ptr::null::<sock_fprog>())).filter as *const _ as usize },
    |                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ this code causes undefined behavior when executed
...
```

Check the bindgen issues if it's already filed. If not prepare a bare min gist and report.
Update `bindgen` once released with a fix.

flxo, 2021-06-23:
Submitted [#2069](https://github.com/rust-lang/rust-bindgen/issues/2069)

inorick, 2021-06-23:
Upstream issue: https://github.com/rust-lang/rust-bindgen/issues/1651

flxo, 2022-04-07:
Closing because the lack of layout tests is acceptable.

closed 2022-04-07 by flxo