esrlabs/northstar #420: Container name and version as parameters to install

csopoldi, 2021-09-05:
The Install API only gets a npk as container image to be installed but no additional information and gets no feedback about the installed container as feedback. 

As a result an application using the Install API call cannot validate what it actually installs to northstar without extracting and parsing the container manifest within the npk.

The Install API call should therefore be extended with container name and version (at least optional) to validate that the npk actually contains the requested container before installing it.

flxo, 2021-09-06:
Hello @csopoldi 

Passing the name and version along is redundant since this information is in the npk. Getting the name an version out of a npk is rather cheap and trivial to accomplish if really needed.

Beside that a application should really know what it passes to the runtime to be installed and not follow the approach to "install and lets see if it's the right thing afterwards".

Adding name and version as the result of a successful installation is a good idea. Feel free to open an issues with a API proposal or even a PR (should be trivial).

cheers,

@flxo

flxo, 2022-04-07:
Fixed by [14302c2](https://github.com/esrlabs/northstar/commit/14302c2b76269bb8f66c349a737aafbf10a98983)