esrlabs/northstar #460: Consider freezing MSRV

jaskij, 2021-12-02:
As an embedded Linux person, seeing MSRV being the current stable version is somewhat undesirable to me. All too often embedded is lagging behind other environments.

Working with Yocto/OpenEmbedded, one of the more popular build systems for custom embedded Linux images, I will not get access to 1.56 (likely 1.58 or 1.59 by then) until the next release in April. Which also happens to be the next LTS release. The current release includes 1.54. This is just one example, I'm sure there are more environments with this issue.

Disclosure: I am not a Northstar user, nor do I plan to, found this repository because of a link in Tokio's Discord. I am simply writing this issue for your consideration.

flxo, 2021-12-02:
@jaskij Thanks for your input. I wasn't aware of the Yocto release cycle duration and haven't heart from the Yocto user of this crate that this has been an issue in the past. They probably just built it in a hacky way out of tree...

Yocto is tested from time to time on `aarch64` from third parties, but is not a prio one target for this project.

The current MSRV is 1.56.0. This is probably missing in the docs but in the CI matrix.

I'll keep this issue until it's at leaset documented.

jaskij, 2021-12-02:
@flxo MSRV is listed in the README, but you have a typo (`MRSV`), so text-search fails. Search for 1.56 instead. (My edits to the issue stem from that).

Rust has been pulled from a separate `meta-rust` into `openembedded-core` in October's Honister (3.4) release, probably due to GNOME's dependency on librsvg, so it may see a wider adoption in the future.

flxo, 2022-01-26:
@jaskij The MSRV is fixed in the README. The CI does a run with the MSRV. I'm closing the issues - feel free to reopen if there's anything left.

closed 2022-01-26 by flxo