esrlabs/northstar #466: Any plans to incubate this to CNCF?

git2g, 2021-12-09:
For increased adoption and the possibility of leveraging [other CNCF projects](https://landscape.cncf.io/), you guys have any plans to incubate this on https://contribute.cncf.io ?

flxo, 2021-12-09:
No. Northstar is targeting "embedded" devices - it's not build for the cloud and totally unsuited for such use cases.

git2g, 2021-12-09:
Well, we were planning to emulate embedded ECUs in the cloud, and the line is blurring these days between "cloud" and "edge". Also looking to see if we could leverage Envoy Proxy to abstract in-vehicle networking. Anyway, there are quite a few CNCF projects that also run in embedded world and vice versa. Northstar is probably one of the few RUST implementations isolating cgroups and namespaces, so it would be good to evolve it for more use cases. Maybe [Linux Foundation](https://www.linuxfoundation.org/projects/hosting/)?

flxo, 2022-04-07:
There are currently no plans. Closing this until further requirements rise.

closed 2022-04-07 by flxo