esrlabs/northstar #474: Use `cargo deny`

adoerr, 2022-01-14:
# Use `cargo-deny`

## What is the feature about ?
[cargo-deny](https://github.com/EmbarkStudios/cargo-deny) is a tool for auditing `Cargo.lock` files for crates with security vulnerabilities, limit the usage of particular dependencies, their licenses, sources to download from, detect multiple versions of same packages in the dependency tree and more.

As such [cargo-deny](https://github.com/EmbarkStudios/cargo-deny)  will provide an **automated** and **tool based** solution to:

- make sure our dependencies only use endorsed OSS licences
- we are aware of [RustSec](https://github.com/RustSec/advisory-db) advisories affecting our code base as well as our dependencies
- we are aware of deprecated dependencies
- keep dependency versions aligned across the dependency tree

## Why is it needed ?

- create initial `deny.toml` file
- adapt default `deny.toml` file, address existing issues
- add GitHub Action for CI integration

## What is expected
Start using [cargo-deny](https://github.com/EmbarkStudios/cargo-deny) for local builds. Once it is properly configured and adapted, integrate into CI

flxo, 2022-01-14:
Thanks.

We have a `cargo-license` check internally but moving to `cargo deny is a good idea` since it checks more stuff than just  the licenses.

adoerr, 2022-01-14:
I will do a PR with an initial `deny.toml` file, so we have something to get going :rocket:

flxo, 2022-01-14:
> I will do a PR with an initial `deny.toml` file, so we have something to get going 🚀

Sorry @adoerr I didn't see this comment before setting up the PR.

PR #481 (2022-01-14): Add cargo deny to CI
Fixes #474
closed 2022-01-14 by flxo