esrlabs/northstar #496: Start using SemVer 'pre-releases'

adoerr, 2022-01-26:
`northstar` has not been released on `crates.io` yet. In addition,  [northstar](https://crates.io/crates/northstar) has already been claimed as a name. Given that, we will likely continue to use `northstar` as a `cargo` git dependency for the foreseeable future.

Currently in order to use a particular version of `northstar`, a `rev` key (with a commit hash) is used together with the `git` key in `Cargo.toml` This is a bit of a nuisance, however.

Since `northstar`, like so many other Rust crates, is still at a `0.x.x` version, using the `version` key is a bit of an issue. Reason is that with a `0.x.x` version, the patch level is basically used to denote minor feature updates and not just patches / bug fixes. On the other hand, `0.6.4` and `0.6.5` are considered compatible from a SemVer point of view. This means that `cargo update` will update a `0.6.4` dependency to `0.6.5`. Which is not what the want, if we are interested in version `0.6.4` in particular.

Starting to use SemVer [pre-releases](https://doc.rust-lang.org/cargo/reference/resolver.html#pre-releases) would help to avoid the above mentioned issue and allow us to use the `version` key. 

Basically a workflow based on `pre-releases` would work like so:

- A version considered "stable" (i.e. it is usable and not WIP) would use a regular SemVer version like say `0.6.4`
- Immediately after version `0.6.4` has been "released" or with the next commit adding / changing code, the version would be set to something like `0.6.5-dev`.

Since from a SemVer point of view version `0.6.4` and `0.6.5-dev` are considered incompatible, a `cargo update` would not touch anything.

Note that I'm aware that using `--precise` together with `cargo update` would allow to "pin"  a dependency to version "0.6.4". I don't think this is a viable alternative, however, because it is a local, manual step and as such quite error prone.

flxo, 2022-01-26:
Thanks for your thoughts. Am I right, that with prereleases a particular version would be equivalent to a git sha? Think so and the advantage would be that versions are easier to handle than shas.  

Northstar is not yet "stable" in any sense but we can try to identify milestones. There's a (not public) long term monkey test running 24/7 which always makes me confident when it doesn't break for a week ;-)

Sadly the name "northstar" is take on crates.io. My boss didn't check that, before choosing Northstar. I personally would like to rename the full crate because it's not used almost anywhere yet. Suggestions are welcome.

adoerr, 2022-01-26:
> Am I right, that with prereleases a particular version would be equivalent to a git sha? Think so and the advantage would be that versions are easier to handle than shas.

Yeah, basically the pre-relase version of, let's say, `0.6.6` is equivalent to the hash of the commit with which the `version` was changed from `0.6.5-dev` to `0.6.6`. So, in the end it is a convenience feature. The issue with using the `rev` key together with a commit hash is that different commit hashes tend to be used in different places. Trying to keep a common version is easier, IMO.

It is understood, that `northstar` is still very much WIP. Using pre-release just makes it a bit more straightforward to stay close to `master` without running into the risk of being 'bitten" by a particular commit.

Regarding renaming, if we want to keep `northstar` for the overall project, the crate itself could be called `polestar` or `nordstern`. Both names are still available.

flxo, 2022-01-26:
> Yeah, basically the pre-relase version of, let's say, `0.6.6` is equivalent to the hash of the commit with which the `version` was changed from `0.6.5-dev` to `0.6.6`. So, in the end it is a [...]

OK. I will try to identify those milestones.

> Regarding renaming, if we want to keep `northstar` for the overall project, the crate itself could be called `polestar` or `nordstern`. Both names are still available.

Hm. Renaming just the crate would have the nice effect that the name `northstar` would be free for the example main that is in the workspace :-). I'll discuss this internally.

TomFleischmann, 2022-03-30:
northstart seem to be free as well. :)

flxo, 2022-04-01:
I really don't like the idea of spamming crates.io with dummy crate names.

flxo, 2022-05-03:
Closing in favour of #555

closed 2022-05-03 by flxo