esrlabs/northstar #632: Please add some functional edits.

lisonok, 2022-05-09:
1. In my opinion, you need to add the ability to connect to the cli of the running container and manage it. `nstar attach`
2. You need to add a command for ndstar to execute a command on a running container. I saw a thread called that, but the code is out of date.

My team and I would like to make a full-fledged analogue of docker in Rust. If someone wants to help us in this difficult matter, please unsubscribe to me in telegram. @TheBeatitude

lisonok, 2022-05-09:
@flxo is it possible to encrypt a container in yuki and prevent other users from viewing its contents?

flxo, 2022-05-09:
@lisonok I have no idea. The youki folks can answer for sure!

flxo, 2022-07-08:
@lisonok I'm closing the issue. Feel free to reopen for more questions or submit another one.

closed 2022-07-08 by flxo