esrlabs/northstar #733: northstar-nstar - README.md is missing

DocMoebiuz, 2022-08-01:
From the main repo there is a link to the nstar specific README.md - but there is no such file:
https://github.com/esrlabs/northstar/blob/main/northstar-nstar/README.md

In general it would be really helpful to have a README.md for each tool in their respective sub-folder. For example, the console tool has "readme"-information but it is located at https://github.com/esrlabs/northstar/blob/main/doc/console.md

Other tools don't have a dedicated README.md at all:
* https://github.com/esrlabs/northstar/tree/main/northstar-client
* https://github.com/esrlabs/northstar/tree/main/northstar-nstar
* https://github.com/esrlabs/northstar/tree/main/northstar-runtime
* https://github.com/esrlabs/northstar/tree/main/northstar-stress
* https://github.com/esrlabs/northstar/tree/main/northstar-test

flxo, 2022-08-03:
Hello @DocMoebiuz 

The link for `nstar` must be fixed. Patches are welcome.

The other crates share the top level readme. Have a look at the cargo manifest where it's referenced.

DocMoebiuz, 2022-08-03:
> Hello @DocMoebiuz
> 
> The link for `nstar` must be fixed. Patches are welcome.
> 
> The other crates share the top level readme. Have a look at the cargo manifest where it's referenced.

@flxo - is the README.md for nstar available in the repository? i can only see the README.md in the root directory, and the one for the sextant tool.

flxo, 2022-08-03:
Not sure. `nstar` ist just a development tool and maybe never had a proper readme. I'll fix the link.

flxo, 2022-08-03:
Fixed by [d370645](https://github.com/esrlabs/northstar/commit/d3706457e30aef2f0ea37e0229b564272a1251d0)

closed 2022-08-03 by flxo