esrlabs/northstar #739: What's the concept to distribute NPKs among a set of targets

DocMoebiuz, 2022-08-02:
I would like to know if there is a concept in place to cover the following user stories:

- As developer I can upload packaged containers (npk) to a central place (registry) to share it with different runtime instances.
- As operator I can define a set of npks and use this definition to update the local registry so that these containers can be started by the instance (e.g. at runtime, or ondemand)

To put it into different words: 
With docker we have central registries where we can push and pull container images from. This is very helpful for sharing containers across teams, workflows, and of course target devices with their respective runtime instances and local registries.

Additionally to being able to have a central place where to store containers, it would also be interesting to know, how to describe and make sure that a set of containers will be started. I would call this container orchestration on the target runtime instance. It kind of like a deployment manifest file, that also could contain information about the required environment variables, that might very well be instance-specific, e.g. ports.

Further thoughts:
* If we wanted to use a ci-workflow to create npks and then store the results, we could use an existing file-oriented registry, like Artifactory OSS.

flxo, 2022-08-03:
A central repository implementation in not in the scope of the runtime. Furthermore keep in mind that containers are not that generic as for example Docker ones are. Every Northstar container is created and configured dedicated for a specific target -> a central repository doesn't make sense.
Updates are specific to the integration of the runtime and can be done by a target specific application.

Regarding orchestration: Northstar targets embedded use cases where a "scripted" orchestration is barely possible or a good choice at all. If you need something like that it's quite easy to realise with a couple of `start` commands with args and env.

DocMoebiuz, 2022-08-03:
Thanks for answering the question and sharing your thoughts.

closed 2022-08-03 by DocMoebiuz