esrlabs/northstar #788: ESR-01-004 WP1: Possible UB due to struct VerityHeader reordering (Info)

flxo, 2022-08-27:


flxo, 2022-09-02:
The struct `VerityHeader` is never used in any FFI context. The block that is append the the verity protected fs is created with [VerityHeader::to_bytes](https://github.com/esrlabs/northstar/blob/e290d9bd42dcc9b5428a208246654b809c7a4b79/northstar-runtime/src/npk/dm_verity.rs#L90). The parsing is done with [VerityHeader::from_bytes](https://github.com/esrlabs/northstar/blob/e290d9bd42dcc9b5428a208246654b809c7a4b79/northstar-runtime/src/npk/dm_verity.rs#L60) (which should be called `from_reader`).

@inorick Can you please double check?

inorick, 2022-09-15:
The `VerityHeader` is not meant to be compatible with the C struct [1]. Reading and writing is done through the `to_bytes` and `from_bytes` methods that take the padding of the C struct into account.

The alternative would be to mark `VerityHeader` as `#[repr(C, packed)]` and add the `_pad1` and `_pad2` members of [1] to it.
However, this would make access to members of this struct unaligned which is deprecated [2]. As I do not see a security issue with the struct and we have no use case for making it more compatible with C, I suggest we keep it as it is.

[1] https://gitlab.com/cryptsetup/cryptsetup/-/wikis/DMVerity#verity-superblock-format
[2] https://github.com/rust-lang/rust/issues/82523

closed 2022-09-15 by flxo

RalfJung, 2022-09-30:
>  this would make access to members of this struct unaligned which is deprecated 

Note that *access* to the fields is fine. Only *references* to those fields are a problem.