esrlabs/northstar #794: ESR-01-012 WP1: Loose Identity Binding of Authentication Tokens (Info)

flxo, 2022-08-27:


inorick, 2022-09-06:
> Since the verifier of a token has to rely on the sender of the token to supply the name that was used to create the container token, the sender has full control over this item.

Who is providing the sender's container name during token verification? Is it provided by the token sender and the target container that wants to verify the token just forwards it?

flxo, 2022-09-06:
The target container just forwards the name but the name is also part of the token right?

inorick, 2022-09-06:
Yes, I think what they mean is that if an attacker compromises container A, obtains tokens from container B and can chose which name is used during verification, he can connect from container A to some target container and claim to be B. If we encode something else into the tokens that cannot be chosen by an attacker when connecting to a target, then the stolen tokens would not be usable that way.

flxo, 2022-10-18:
@inorick How do we continue here? As far as I get it this issue cannot be solved by Northstar because it must be supported by the protocol/transport mechanism in use e.g unix socket credentials.

inorick, 2022-10-19:
Agreed. The problem is that the token is tied to its creator and not to the sender of a particular connection. To achieve this, we would need a way for the target container to ask the runtime: "What container is this connection really originating from". Then it could use that information and compare it to the claimed identity of the sender. I do not think that there is anything we could add to a token to tie to a connection (like an IP address) that could not be spoofed by an attacker.

As a side note, I do not think that this problem would go away if we used a full blown PKI. An attacker that obtains the private key of one container can use it to impersonate that container from another container.

inorick, 2022-10-24:
I think we can close this as I do not see any unspoofable connection identifier that we could tie to the tokens.

flxo, 2022-10-24:
@inorick Agree. This can only be fixed with support from the protocol or transport used. The runtime cannot.

closed 2022-10-24 by flxo

1ohn-Brown1, 2023-05-05:
To the people who stole and renamed MY code and MY encryption from ÆTHOR to the multiple different things you call it now....the real author [TH] is coming for you! And soon you'll see just how automagically all my identifiers are restored when I take the hard copy proof I have of your theft to court! I bet you never did find the master LOL....PG 105 is altered because of a fail safe I programmed in that printed a specific word and picture  upon the theft and alterations of my my proprietary property. MRÆ is me CI McGee. You should have checked the FIG! Ccc u sn

flxo, 2023-05-05:
wow. Github spam...