esrlabs/northstar #913: Python Applications in Containers

Hamzaslipax, 2023-02-03:
Hello, I have some issues trying to start Python Apps in containers. I can't exactly know what the issue is if the container need more resources or more permissions. I created the Executable using PyInstaller.

2023-02-03T13:32:09.807               runtime::state ⬤  INFO : Started grpc-python-server:0.0.1 (95688) in 0.024s
2023-02-03T13:32:09.807     grpc-python-server:0.0.1 ⬤  DEBUG: 2023-02-03T13:32:09.807            runtime::fork::util ⬤  DEBUG: Setting parent death signal to SIGKILL
2023-02-03T13:32:09.808             runtime::console ⬤  INFO : Remote(tcp://127.0.0.1:39712): Connection closed
2023-02-03T13:32:09.817     grpc-python-server:0.0.1 ⬤  **DEBUG: [2] INTERNAL ERROR: cannot create temporary directory!**

I would be so happy if someone can help.
Thank you!

flxo, 2023-02-03:
Looks like your applications tries to create "some temporary directory". Probably the location it tries to access is not writeable. You can verify by enabling `strace` in the runtime configuration and you're capable of reading `strace` output.

try to add 

```yaml
  /tmp:
    type: tmpfs
    size: 10MB
``` 

to your container mounts. This will add a writeable `tempfs` directory `/tmp`. Just a guess :-)

Hamzaslipax, 2023-02-03:
Thank you for your response!

After trying to add /tmp to my mounts it looks better but still missing something. I guess I need to remount /tmp.

```
2023-02-03T15:59:58.811               runtime::state ⬤  INFO : Started python:0.0.1 (107744) in 0.017s
2023-02-03T15:59:58.811                 python:0.0.1 ⬤  DEBUG: 2023-02-03T15:59:58.811            runtime::fork::util ⬤  DEBUG: Setting parent death signal to SIGKILL
2023-02-03T15:59:58.812             runtime::console ⬤  INFO : Remote(tcp://127.0.0.1:55300): Connection closed
2023-02-03T15:59:58.935                 python:0.0.1 ⬤  DEBUG: /python: error while loading shared libraries: libz.so.1: failed to map segment from shared object
```

closed 2023-02-05 by Hamzaslipax

flxo, 2023-02-06:
What do you mean by remount? Your container probably doesn't have the capabilities to remount anything.

Regarding the so load error I'd have a look a `strace`.