esrlabs/northstar #917: IPC Namespaces

Hamzaslipax, 2023-02-09:
Northstar automatically creates an IPC Namespace for each container, unlike Network Namespace that require the user to preconfigure before a container can join it.

My question: The IPCs such as shared memory can then only be used within the container. Is it possible for many containers to join the same IPC namespace to allow IPC between containers?
I am writing my bachelor thesis about Northstar, I would appreciate if someone could help.

flxo, 2023-02-09:
Hello @Hamzaslipax 

Currently a container cannot join an *existing* IPC namespace. This would be probably easy to add as it is done for the network namespaces. This features requires an manifest update to add the namespace to join etc... PRs are always welcome :-). I'll add an issue for the feature.

cheers,

Felix

ps: I'll ping you regarding your thesis.

Hamzaslipax, 2023-02-09:
Hello Felix, Thank you so much for the clarification!
I would be happy to share it when I'm done ^^.

Best regards,
Hamza