esrlabs/northstar #971: How can I run strace on a app that crashs at startup?

shaeussler, 2023-05-09:
Hello, 

I'm using the v0.7.1 and I have a app that crashs during startup because of a missing dependency. 
I want to use strace but I need the [init PID](https://github.com/esrlabs/northstar/blob/07455e6fff98f3f87a3753425a75537d54922925/northstar.toml#L37). 
How can I find out the PID of the init-process of my app dynamically? 
Is it possible to have a fixed init PID?
I tried "sudo strace -f -s 256 -p `pidof <container name>`" but it did not work.

I would be so happy if someone can help.
Thank you!

flxo, 2023-05-10:
> I'm using the v0.7.1 and I have a app that crashs during startup because of a missing dependency. I want to use strace but I need the [init PID](https://github.com/esrlabs/northstar/blob/07455e6fff98f3f87a3753425a75537d54922925/northstar.toml#L37). 

`<PID>` is the *init* pid.

shaeussler, 2023-05-10:
Hi @flxo ,
Thanks for your help. 
I will try the strace in the init.

closed 2023-05-10 by shaeussler