esrlabs/northstar #976: How to mount TPM device?

shaeussler, 2023-05-16:
Hi,

I tried to mount a TPM device with the following manifest:
```
mounts:
    /dev:
        type: dev
    /dev/tpm0:
        type: dev
    /dev/tpmrm0:
        type: dev
```
Inside the container the TPM devices are directories and no charachter devices?
[image]

How can I mount the TPM devices as charachter devices?
I also tried the mount type bind for the TPM but get an nothstar "ENODIR" panic error at container startup.

I would be so happy if someone can help.
Thank you!

flxo, 2023-07-17:
Hello @shaeussler 

Sorry for the late reply - I've been on vacation for a longer period :-)

The type `dev` of a mount is a special type. Northstar creates a minimal set of device files typically needed e.g `stdout`, `null` etc.... 
You *cannot* mount file via the manifest. You're trying that for `tpm0` and `tprm0` and this will fail.

So - you need the `/dev` from your host which is currently possible via a simple bind mount including the `rw` option.

```yaml
mounts:
  /dev:
    type: bind
    host: /dev
    options: rw
```   

cheers,

@flxo

shaeussler, 2023-07-17:
@flxo Thanks for the response. I will test it.

closed 2023-07-17 by shaeussler